yujinrobot/kobuki #353: Scan_angle issue in gyro_perf.py

ElizavetaElagina, 2015-02-09:
Hello!
On launching gyro_perf.py I get the following error: 'Tester' object has no attribute scan_angle. What do I do with it?

Best regards,
Elizaveta

slivingston, 2015-02-16:
The error appears to be due to a race condition between `timeCallback()` (registered on [line 69](https://github.com/yujinrobot/kobuki/blob/0.6.5/kobuki_testsuite/scripts/gyro_perf.py#L69)) and `angleCallback()` (registered on [line 60](https://github.com/yujinrobot/kobuki/blob/0.6.5/kobuki_testsuite/scripts/gyro_perf.py#L60)).

slivingston, 2015-02-16:
There are a variety of ways to fix this. I am not sure what is preferred upstream, but you could try
```diff
@@ -49,6 +49,8 @@ class Tester(object):
     self.diff_angle = 0.0
     self.accumulated_angle = 0.0
+    self.scan_angle = None  # Place holder until the first scan angle is received

     self.init_time = rospy.Time.now().to_time()
     if self.debug: print 'init time: {0:2.4f}'.format(self.init_time)
 
@@ -96,6 +98,8 @@ class Tester(object):
         return
 
     if self.state == "ALIGNING":
+      if self.scan_angle is None:
+        return
       scan_angle = self.scan_angle
       if abs(scan_angle) > radians(1.0):
         cmd_vel = Twist()
```

slivingston, 2015-02-16:
@ElizavetaElagina, if you are interested in studying drift on the Kobuki in general rather than using gyro_perf.py in particular, then you should consider `drift_estimation` in the `kobuki_testsuite` Python package (under `kobuki_testsuite/src/kobuki_testsuite/` in this repository). A GUI front-end is available in [kobuki_qtestsuite](http://wiki.ros.org/kobuki_qtestsuite).

jihoonl, 2015-02-17:
@ElizavetaElagina 

I have patched the script as @slivingston suggested. Let me know if you still see a problem.

@slivingston Thanks for patch!

closed 2015-02-17 by jihoonl

ElizavetaElagina, 2015-02-23:
@jihoonl @slivingston Thank you very much! It works fine now.