yujinrobot/kobuki #354: Auto_docking and move_base_nodelet_manager fail when the dock station isn't found

ferherranz, 2015-04-17:
Development environment : Ubuntu 14.04 / ROS Indigo
I start the auto_docking node when the dock_station isn't visible from the kobuki's position

1. roslaunch kobuki_node minimal.launch --screen
2. roslaunch kobuki_auto_docking minimal.launch --screen 
3. roslaunch kobuki_auto_docking activate.launch --screen 

Then, the move_base_nodelet_manager crashes and shows the following message:
 
[mobile_base_nodelet_manager-2] process has died [pid 23130, exit code -11, cmd /opt/ros/indigo/lib/nodelet/nodelet manager __name:=mobile_base_nodelet_manager __log:=/home/user/.ros/log/c87fa0de-e4ef-11e4-b504-a0a8cd4bec86/mobile_base_nodelet_manager-2.log].
log file: /home/user/.ros/log/c87fa0de-e4ef-11e4-b504-a0a8cd4bec86/mobile_base_nodelet_manager-2*.log

The  Auto docking nodelet shows:

[ INFO] [1429267952.227301129]: Bond broken, exiting
[dock_drive-1] process has finished cleanly

and the DockDriveActionClient shows the following messages, and keeps waiting:
...
Feedback: [DockDrive: SCAN]: 
Feedback: [DockDrive: SCAN]: 
Feedback: [DockDrive: SCAN]: 
Feedback: [DockDrive: SCAN]: 
Feedback: [DockDrive: SCAN]: 
Feedback: [DockDrive: FIND_STREAM]:

henoSH, 2015-10-12:
Halllo i have the same problem. Did you find a solution?

ferherranz, 2015-10-26:
I couldn't work on it, sorry

henoSH, 2015-10-27:
thank your for your mail. but i want to ask did you do some own project
using Turtlebot?

magehrig, 2015-12-12:
I can confirm this issue. I have not found any solution.

henoSH, 2015-12-14:
Problem solved verify if the IR receiver in the middle has data or no.

tinokl, 2016-05-09:
Got the same problem and fixed it by modifing:
Inside the dock_drive_cpp of kobuki core pkgs (https://github.com/yujinrobot/kobuki_core)

Line 121:   std::string debug_str; 
to 
Line 121:   std::string debug_str = "";

hope this helps

stonier, 2016-05-10:
As far as I know, the default `ctor` of a string creates an empty string which should have exactly the same effect as creating it with `""` ([cplusplus strings](http://www.cplusplus.com/reference/string/string/string/)). Using the default ctor instead of explicitly assigning it *is* the preferred way.

So I suspect something deeper is going wrong.

tinokl, 2016-05-17:
true story... what i did, was removing the nodelet manager stuff which made debugging a little confusing and started the docking in an extra node. 

Here i used valgrind and the debug string was one failure then it looked quite fine (when running)
The funny fact is, the node keeps restarting with no error or what so ever and then the goal has to be resend. 

When doing so, everything is turning out well (except that the node is restarting and the "roslaunch kobuki_auto_docking activate.launch --screen" client script is not able to deal with an empty status and crashes when receiving one)

tinokl, 2016-05-19:
so here is what i did and what i found :)
again i tried a little with valgrind and so forth and got message filter errors and after removing the filter i got the real problem...

Everything is inside kobuki_core/kobuki_dock_drive/src/dock_drive.cpp.

the variable "new_state" was wrongly set. thus i get an error on picking the robot state string here:
"state_str = ROBOT_STATE_STR[new_state];" at row 280
after investigating i found several problems and that solved the problem so far^^:

# 1: in "void DockDrive::find_stream(RobotDocki..." the state "RobotDockingState::State next_state"
is never initiated if the robot is neither left nor right the dock. 
solved by changing the line: 
"else if(dock_detector < 0 ) // robot is located in left side of dock" 
to 
 "else if(dock_detector <= 0 ) // robot is located in left side of dock

# 2: same as in #1 but in "get_stream(...)"

# 3: not allways a bug, depending on compiler is following line in "updateVelocity"
 "new_state = current_state = state;"
Some compiler could interpret this line different then intended.
better and of course saver is just:
  "new_state = state;
  current_state = new_state;"

hope this is the final problem :)
kind regards