yujinrobot/kobuki #358: Question: maximum velocity of Kobuki

jvgomez, 2015-06-02:
Hi, I see in the [Kobuki specifications](http://kobuki.yujinrobot.com/home-en/about/specifications/) that the maximum linear speed is 0.7 m/s. During teleoperation, the maximum seems to be 1.5 m/s. However, when giving velocity commands from a node the maximum it can reach is 0.5 m/s. Where is this configured and what is the real maximum velocity of the platform?

Thanks!

stonier, 2015-06-03:
Are you using a kobuki based turtlebot when giving velocity commands or just a `kobuki_node` running on top?

I don't think the basic kobuki launchers generate any limitations, but kobuki `safe_keyop.launch` and the default `turtlebot_bringup` launchers attach a velocity smoother to incoming velocity commands. These have parameters for velocity and acceleration bounds. This is probably what's doing it.

* `kobuki_keyop`/param/keyop_smoother.yaml`
* `turtlebot_bringup/param/defaults/smoother.yaml`

We do have some acceleration profile pictures we made when developing the kobuki...somewhere. But we never posted these online. You can generate these by modifying/running some of the scripts in `kobuki_testsuite/scripts` and viewing results in `rqt_plot`.

jvgomez, 2015-06-03:
I'm using the `kobuki_node`. When I launch `roslaunch kobuki_keyop keyop.launch` I can clearly see what the limits are, as they are in a yaml file.  However, when I publish `cmd_vel` commands it does not get over 0.5 m/s in simulation (I haven't tested in the real robot yet). I am investigating further, but what I am 99% sure is that it is not the `keyop_smoother.yaml`. I will post further information soon.

bit-pirate, 2015-06-04:
@jvgomez what simulation are you using? Gazebo?

jvgomez, 2015-06-04:
@bit-pirate Yes, I am using Gazebo.

jvgomez, 2015-06-05:
For some reason I am now able to give a maximum velocity of 0.7 m/s in the real robot, while velocity in simulation is not limited (or given as parameters). So I close this issue as the question is solved. Thank you!

closed 2015-06-05 by jvgomez

bit-pirate, 2015-06-05:
Would still be interesting, where the simulation is actually limiting the velocity.

jvgomez, 2015-06-05:
It should be because of my code, because if I give a manual velocity reference the robot is able to follow it.