yujinrobot/kobuki #359: "gyro_link" in the imu_data header should not be hard coded but should be resolved with tf_prefix

quattrinili, 2015-08-19:
In slot_callbacks.cpp of kobuki_node at line 126 and at line 167 the frame of the gyro shouldn't be hard coded (i.e., msg->header.frame_id = "gyro_link";) but it should be resolved with the tf_prefix, otherwise it is not possible to seamlessly use the imu_data with robot_pose_ekf in case there are multiple robots with different tf_prefixes.

bit-pirate, 2015-08-20:
> In slot_callbacks.cpp of kobuki_node at line 126 and at line 167

Next time please with link: https://github.com/yujinrobot/kobuki/blob/indigo/kobuki_node/src/library/slot_callbacks.cpp#L126

Makes it easier to check it out. :-)

> but it should be resolved with the tf_prefix,

Working on this would probably mean a bigger change, e.g.

* where to get the robot's namespace from?
* tf_prefix is being deprecated with TF2 (http://wiki.ros.org/tf2/Migration#Removal_of_support_for_tf_prefix). What would be the proper way to implement this?
* What other parts of Kobuki (and TurtleBot) need to be changed?

If you have an idea how to overcome these challenges, we would be happy to review a PR.

quattrinili, 2015-09-23:
Sorry for my late reply, but I had a deadline and so I was very busy.
Below, my answers to some of your points.

> In slot_callbacks.cpp of kobuki_node at line 126 and at line 167
>
> Next time please with link:
> https://github.com/yujinrobot/kobuki/blob/indigo/kobuki_node/src/library/slot_callbacks.cpp#L126
>
> Makes it easier to check it out. :-)

Yeah, thanks, next time I will point directly to the link :-)

> but it should be resolved with the tf_prefix,
>
> Working on this would probably mean a bigger change, e.g.
>
>    - where to get the robot's namespace from?
>
> In the short term, an easy fix could be to directly resolve resolve the
transform with the tf_prefix -- i.e.,
TransformListener.resolve("gyro_link"), which should automatically retrieve
the tf_prefix set by a user in the launch file.

>
>    - tf_prefix is being deprecated with TF2 (
>    http://wiki.ros.org/tf2/Migration#Removal_of_support_for_tf_prefix).
>    What would be the proper way to implement this?
>
> Indeed I know that it could be a bigger change, especially considering
that at some point projects will migrate to TF2, even if I don't have a
straightforward answer to this question.

>
>    - What other parts of Kobuki (and TurtleBot) need to be changed?
>
> Regarding the hardcoded transforms, some other parts of the code that I
think need to be changed (e.g., by adding TransformListener.resolve) are:
-
https://github.com/yujinrobot/kobuki/blob/73df34f9ce7bb600909943a53c41cb9c2f735e43/kobuki_node/src/library/odometry.cpp#L39
-
https://github.com/yujinrobot/kobuki/blob/73df34f9ce7bb600909943a53c41cb9c2f735e43/kobuki_bumper2pc/src/kobuki_bumper2pc.cpp#L93