yujinrobot/kobuki #365: I got an error because I just got my Kobuki.

AyanGithub, 2016-06-21:
I just got my Kobuki a few days ago so I was happy to play with it. But then I got an error while trying to catkin_make the Kobuki workspace within Turtlebot. The error was:
```
CMake Error at /home/ayan/turtlebot/devel/share/ecl_exceptions/cmake/ecl_exceptionsConfig.cmake:141 (message):
  Project 'kobuki_keyop' tried to find library 'ecl_exceptions'.  The library
  is neither a target nor built/installed properly.  Did you compile project
  'ecl_exceptions'? Did you find_package() it before the subdirectory
  containing its code is included?

```
Any help is appreciated!

closed 2017-03-23 by AyanGithub

lukelu0520, 2020-08-03:
@AyanGithub Got the same issue, was wondering how to resolve the issue.

stonier, 2020-08-03:
One rosdep update call on this repo in your source workspace should be all it needs, that will track down all the dependencies you need and install them via apt.