yujinrobot/kobuki #367: Modification of gyro rate

Severn-Vergil, 2016-08-14:
Can the L3G4200D gyro rate be modified in the driver of ROS? The gyro update rate in Kobuki is approximately 104Hz(I measure it by rostopic hz), however the highest rate is up to 855hz, so I wonder if the gyro data is supsampled. A higher sample rate of gyro is needed in my application.
Thank you in advance! Your advice would be great help to me.

stonier, 2016-09-20:
I don't have full knowledge of what is in the firmware on the arm chip, but I'd say that is the constraint there. The pc driver certainly doesn't provide any parameterisation to access the gyro data at a higher rate and since the firmware blob is closed, there would have to be some development to provide access to the gyro data at a higher rate.

Unfortunately we're not actively developing the firmware at this time.

Severn-Vergil, 2016-09-20:
Thank you @stonier, is there any plan for releasing a new version of Kobuki?

stonier, 2016-09-20:
Not a new version of kobuki per se. There is a plan to put ROS2 on kobuki though and that might allow development time on the firmware.