yujinrobot/kobuki #371: Is my kobuki's odom   accurate?

hapi007, 2016-10-12:
I have tested my kobuki's odom using " rosmsg show  nav_msgs/Odometry" AND  spin my kobuki 10 times,keeping  it's oriention coincident with the begining to my best .    
I found that the msg.pose.orientation  changed from 0 to 20 degree(i used math tools to compute it).
I  don't know  this result is  reasonable or not.

stonier, 2016-10-26:
Shouldn't be anywhere near that bad. A couple of degrees a minute at worst. Make sure the robot is on the ground and not moving when it is turned on. Also make sure it is not rotating on an angle from the horizontal plane. The speed of rotation can have some influence as well - how fast do you spin?

hapi007, 2016-10-27:
I used ” roslaunch turtlebot_teleop keyboard_teleop.launch ” to test it，used key ‘J’ and ‘L’ to control。And the speed of rotation is default：
currently:      speed 0.2       turn 1 
I am also sure my kobuki is on the horizontal plane。I spin it 10times，it spend about 1minute。
At the beginning，the odom pose is：
pose: 
  pose: 
    position: 
      x: -0.0415894216405
      y: -0.000834579950722
      z: 0.0
    orientation: 
      x: 0.0
      y: 0.0
      z: 0.0474551261763
      w: 0.998873370853

After rotate 10 times，the pose is：

pose: 
  pose: 
    position: 
      x: -0.0397655645054
      y: 0.000896180439553
      z: 0.0
    orientation: 
      x: 0.0
      y: 0.0
      z: -0.143492621991
      w: 0.98965138682