yujinrobot/kobuki #382: Malformed sub-payload detected.

BasJ93, 2018-02-01:
I am running the Turtlebot2 with the Kobuki base. Both on Indigo en Kinetic, I receive the error:
`Kobuki: malformed sub-payload detected, ...`

The sub payload that is shown is always a sub-payload of the base-state as defined by the protocol specification with a lenght of 77 bytes. (`AA 55 4D 01 0F`) However, the reported error does not have 77 bytes of data. This does match the behaviour as described in kobuki.cpp line 405 of the kobuki_driver.

I have tested this with two laptops running in Indigo and one running on Kinetic, and two different Kobuki bases.

mzaiady, 2018-02-07:
I'm facing the same issue, using Intel NUC, it was working fine before! but this this error start to show-up!
 
```
[ERROR] Kobuki : malformed sub-payload detected. [225][170][E1 AA 55 4D 01 0F ]
[ERROR] Kobuki : malformed sub-payload detected. [42][170][2A AA 55 4D 01 0F ]
[ERROR] Kobuki : malformed sub-payload detected. [94][170][5E AA 55 4D 01 0F ]
[ERROR] Kobuki : malformed sub-payload detected. [63][170][3F AA 55 4D 01 0F C0 E8 00 00 00 ]
[ERROR] Kobuki : malformed sub-payload detected. [221][170][DD AA 55 4D 01 0F 78 08 00 00 00 94 ]
[ERROR] Kobuki : malformed sub-payload detected. [43][170][2B AA 55 4D 01 0F 24 0C 00 00 00 ]
[ERROR] Kobuki : malformed sub-payload detected. [103][170][67 AA 55 4D 01 0F F0 80 00 00 00 94 ]
[ERROR] Kobuki : malformed sub-payload detected. [143][170][8F AA 55 4D 01 0F 30 8C 00 00 ]
[ERROR] Kobuki : malformed sub-payload detected. [27][170][1B AA 55 4D 01 0F 80 9B 00 00 00 94 58 63 9C 00 00 00 06 91 00 ]
[ERROR] Kobuki : malformed sub-payload detected. [62][170][3E AA 55 4D 01 0F 0C ]
[ERROR] Kobuki : malformed sub-payload detected. [212][170][D4 AA 55 4D 01 0F ]
[ERROR] Kobuki : malformed sub-payload detected. [193][170][C1 AA 55 4D 01 0F 18 30 00 00 00 ]
[ERROR] Kobuki : malformed sub-payload detected. [230][170][E6 AA 55 4D 01 0F AC DC 00 00 00 94 ]

```

ROS Indigo,
Ubuntu 14.04.5 LTS
kernal 4.4.0-112-generic

```
 rosrun kobuki_driver version_info
Version Info:
 * Hardware Version: 1.0.4
 * Firmware Version: 1.2.0
 * Software Version: 0.6.5

```

lukelu0520, 2020-08-05:
@BasJ93 @mzaiady , got the same issue with you in 2020. Do you find any solution for the error message "Malformed sub-payload detected" ? For example, update firmware or software ?

Hope to hear your expertise.

-- Luke

stonier, 2020-08-05:
Does it show up continuously or sporadically? Is it an old kobuki?

lukelu0520, 2020-08-05:
Hi @stonier ,

It shows up when I execute: roslaunch kobuki_node minimal.launch --screen
I just purchased and received it from Trossen Robotics, don't know if it is old or not, checked the version which shows:

Version Info:
 * Hardware Version: 1.0.4
 * Firmware Version: 1.2.0
 * Software Version: 0.7.8

Hope to hear your expertise on the error message.

-- Luke

stonier, 2020-08-05:
The only thing that stands out for me (given that software/firmware works in general and has suddenly gone from working to not working) is that noise on the serial connection interferes with the payload. I've seen this happen on serial connections that are long / old / have a great deal of electromagnetic interference. Haven't seen it on a kobuki yet. 

To debug further, see if you can isolate the problem to the communication line. Some experiments:

* Try a different cable (new and shorter is ideal for comparison purposes) and see if that helps?
* Get a laptop with serial port and use the serial comm device directly (not the usb).

lukelu0520, 2020-08-05:
> The only thing that stands out for me (given that software/firmware works in general and has suddenly gone from working to not working) is that noise on the serial connection interferes with the [...]

Thanks for your advice @stonier , do you have any recommended serial communication cable which is short enough and interference free for the test ?  In the same time, I'll get few shorter and better usb cable for further testing.

-- Luke

stonier, 2020-08-05:
Nothing special. You're just chasing some second opinions on the communication problem to verify it isn't the cable that you have that is the source of the problem.

stonier, 2020-08-05:
Note - you shouldn't have a software/firmware mismatch. If there was a major version mismatch, it would immediately abort and if there was a minor version mismatch you'd get a warning on startup. See [here](https://github.com/kobuki-base/kobuki_core/blob/devel/kobuki_driver/src/driver/kobuki.cpp#L298).

lukelu0520, 2020-08-05:
Thanks for your prompt reply @stonier 

BTW, what does "payload" and "sub-payload" in the context mean ? Is it a mechanical fixture to support physical loading or a virtual digits in protocols for communication ?

stonier, 2020-08-05:
Each packet carries a variable length chain of data packets depending on what was requested in the last command window. Each sub-payload caries the data for a single quantity (e.g. cliff sensors). You can see details (not great english, but you'll get the idea) here: http://yujinrobot.github.io/kobuki/enAppendixProtocolSpecification.html

lukelu0520, 2020-08-12:
Thanks for your suggestion @stonier 
Now we can eliminate cable issue since we've tried multiple cables with a variety of length (from 0.5 feet to 3 feet).
Still figure out the root cause of malformed sub-payload detected, will check if there is a ros way to see the full packet stream.

-- Luke

corot, 2020-08-13:
@lukelu0520, do you have several USB devices connected to the same hub (or same USB board within the PC) as Kobuki? I noticed with my robot that the number of `malformed sub-payload detected` errors increases a lot when that happen.

swinterbotix, 2020-08-13:
I think I fixed this issue! By default the usb latency timer in Linux is set to 16ms for FTDI usb communication. By changing this to the lowest possible value (1ms), this issue disappears! Let me know if this works for you guys and I'll submit a PR to update the udev rules for kobuki.

There are a couple ways to change the latency timer. The best approach is to just edit the kobuki udev rule. Mine is located at /lib/udev/rules.d/60-ros-melodic-kobuki-ftdi.rules.

Substitute the line with ...

`SUBSYSTEM=="tty", ATTRS{idVendor}=="0403", ATTRS{idProduct}=="6001", ATTRS{serial}=="kobuki*", ATTR{device/latency_timer}="1", MODE:="0666", GROUP:="dialout", SYMLINK+="kobuki"`

stonier, 2020-08-14:
Oh, looks like the world shifted in linux kernel 4.10 (which was after the initial releases of kobuki) for the ftdi driver. See [here](https://github.com/ros-drivers/rosserial/issues/383) and [here](https://lore.kernel.org/patchwork/patch/763463/). That would explain why the problems are occurring now and not when Kobuki was initially released.

That udev rule could well work (looking forward to hearing from others), but perhaps a better solution is to do it programatically (see the second link above) so the `/dev/kobuki` port isn't spinning insanely, even when idle.

stonier, 2020-08-14:
A long thread over in Octopi about this: https://github.com/guysoft/OctoPi/issues/566

Interesting to note, if we could pad packets out to multiples of 64bytes, it would fire off the chain of sub-packets with no 16ms delays. Unfortunately we don't have access to the firmware :/

Potentially, some code that could set the low latency flag:

```
# serial struct is defined in <linux/serial.h>
ioctl(file_descriptor, TIOCGSERIAL, &serial); 
serial.flags |= ASYNC_LOW_LATENCY; // (0x2000)
ioctl(file_descriptor, TIOCSSERIAL, &serial);
```

However, folks in threads I've read mention that you need root permissions to set that flag :/

swinterbotix, 2020-08-14:
> That udev rule could well work (looking forward to hearing from others), but perhaps a better solution is to do it programatically (see the second link above) so the /dev/kobuki port isn't spinning insanely, even when idle.

Correct me if I'm wrong (and there's a good chance I am since I don't really work in serial communication) but it sounds like the latency timer only comes in to play when a 'read' command is sent to a device from a host computer. If no 'read' command is sent, which I imagine would be the case if the Kobuki is just plugged in to the computer but no code is running, then there wouldn't be an issue?

stonier, 2020-08-20:
Not an expert either, but looks like you're correct. Reference: [FTDI Knowledgebase - Custom Latency Timer Value](https://www.ftdichip.com/Support/Knowledgebase/index.html?settingacustomdefaultlaten.htm).

Having said that - this looks like the right thing to do, if only to reduce the latencies in reading the packet stream. That still doesn't explain however why the data is malformed at 16ms latencies, I would have expected it to be ok, just slower on the uptake.

stonier, 2020-08-20:
For reference, you can determine the configured latency timer on your loaded device with:

```
cat /sys/bus/usb-serial/devices/ttyUSB0/tty/ttyUSB0/device/latency_timer
```

I can confirm the udev rule above does set it to 1ms.

I'm getting some interesting results with [stream.cpp](https://github.com/kobuki-base/kobuki_core/blob/devel/kobuki_driver/src/demos/stream.cpp):

With 16ms latency:

```
[879570.639372021] Encoders [65532,65517]
[879570.697347595] Encoders [65532,65517]
[879570.720376262] Encoders [65532,65517]
[879570.779421650] Encoders [65532,65517]
[879570.837478482] Encoders [65532,65517]
[879570.860657557] Encoders [65532,65517]
[879570.920656717] Encoders [65532,65517]
[879570.978353900] Encoders [65532,65517]
[879571.001313090] Encoders [65532,65517]
[879571.060234311] Encoders [65532,65517]
[879571.118218788] Encoders [65532,65517]
[879571.140584549] Encoders [65532,65517]
[879571.200559087] Encoders [65532,65517]
```

With 1ms latency:

```
[880250.400911652] Encoders [65532,65515]
[880250.421051170] Encoders [65532,65515]
[880250.440984560] Encoders [65532,65515]
[880250.461133893] Encoders [65532,65515]
[880250.482130926] Encoders [65532,65515]
[880250.500816462] Encoders [65532,65515]
[880250.521110288] Encoders [65532,65515]
[880250.541922923] Encoders [65532,65515]
[880250.561056298] Encoders [65532,65515]
[880250.580858288] Encoders [65532,65515]
[880250.600774594] Encoders [65532,65515]
[880250.621897606] Encoders [65532,65515]
[880250.641907712] Encoders [65532,65515]
[880250.661814168] Encoders [65532,65515]
[880250.680774966] Encoders [65532,65515]
[880250.701015174] Encoders [65532,65515]
```

Conclusions:

* At 16ms, packets arrive @ 15-60ms with high variance
* At 1ms, packets arrive @ 20ms, with minimal variance <- this was the behaviour when kobuki was released (and of course, the desirable behaviour!)

I don't seen any negative ramifications due to increasing the polling rate that affects usual use of kobuki.

+1 for getting this patch in.

lukelu0520, 2020-08-21:
@swinterbotix @stonier , amazing, problem solved after adapting your solution.
Super happy we get rid of the error message !

swinterbotix, 2020-09-09:
Do you plan to make version 0.7.9 available on APT for kinetic/melodic? It's still showing 0.7.8 as the most recent version.

stonier, 2020-09-10:
http://repositories.ros.org/status_page/ros_melodic_default.html?q=kobuki

You'll see that it's in the staging area, but hasn't been sync'd to the release servers. You can get access to it via the ros-testing server, see [this wiki page](http://wiki.ros.org/TestingRepository). 

I didn't release it in kinetic. I can't test it at the moment and there are some changes that might not backport so was cautious about sync'ing branches. If you would like to use it there, and can confirm the udev rule changes in 45f834cd work, I'll cherry-pick that single change and re-release.

stonier, 2020-09-14:
On it's way in kinetic, melodic & noetic, `v0.7.10`.

closed 2020-09-14 by stonier

corot, 2020-12-31:
Finally regain access to my old, good kobuki;  
Malformed sub-payload errors gone!!!  thank u guys,, and happy new year!