yujinrobot/kobuki #386: i can't find the left/right wheel topic on turtlebot Kobuki ?

NabilRez, 2018-03-06:
Hi, the "/cmd_vel_mux/input/teleop" topic controls linear velocity and angular velocity of the turtlebot kobuki,
What i am trying to find is the topic that controls the rotational speed of the left/right wheel of the turtlebot kobuki in order to implement an LQ regulator to control the speed loop of the turtlebot.

stonier, 2018-03-13:
Been a while, but I think that that pair is what routes directly into the protocol to the Kobuki ECU. You should be able to compute single wheel rotational speeds and then transform them into the above form (base linear velocity/rotational velocity) using the Kobuki base parameters (wheel diameter, ...). Likely to be some error coming from the base parameters, but it should be fairly negligible.

NabilRez, 2018-03-14:
Ok i have some question ( please have in mind that i am new at ROS ).
what is Kobuki ECU ? ( or simply point me to where i can get more understanding about kobuki ecu )
also, in order to implement different control strategy to the speed loop of the turtlebot, two things are required :
                  -Publish into the topic that controls the (left / right )  wheel rotational  speed of the turtlebot.
                  -Get the measure of the actual rotational speed of the (left/right) wheel form the turtlebot, either by directly subscribing from the wheel encoder topic, which is more practical, or infer it from the speed of the turtlebot (i.e: deduce the ( left / right ) wheel rotational speed from the actual speed of the turtlebot via an inverse matrix )

alexantosh, 2020-08-20:
hello, were you able to find this out? How to control the different wheels separately?

stonier, 2020-08-23:
Seems like something is wrong - you should be able to perfectly manage that with the control paradigm the ros1 node has. I haven't looked / tested the ros1 code for years (and unable to check on melodic right now) so can't help you out there right now.

However, you could try to compile and run the simple loop demo (no ros1 involved and that api will work against melodic's c++ kobuki_driver library). Source code is here.

https://github.com/kobuki-base/kobuki_core/blob/devel/kobuki_driver/src/demos/simple_loop.cpp

Just copy/paste it into your build environment somewhere. It controls kobuki around a virtual square, with pure rotations of the sort you want at the corners. This will at least tell you whether there's a problem with your kobuki/kobuki driver or it's some bug in the ros node wrapper around it.

alexantosh, 2020-08-24:
ok i will try this. No, the turtlebot2 and kobuki are running on ROS Indigo Ubuntu 14.04. Couldnot find a way to handle that with ROS. I practically send a Twist Object with just the angular velocity and it seems only one wheel is turning and somehow the robot displaces. In ROS melodic I have gazebo. Thats where I want to first try this then port it to the hardware.

alexantosh, 2020-08-24:
I tried the code sample you gave me on both ubuntu 18.04 and 14.04 but it gives errors. 
for Ubuntu 14.04, when i try to compile the file, i get:

`turtle1@turtle1:~/kobuki/src/kobuki_core/kobuki_driver/src/test$ gcc -o sim simple_loop.cpp 
In file included from /opt/ros/indigo/include/ecl/linear_algebra.hpp:19:0,
                 from /opt/ros/indigo/include/ecl/geometry/legacy_pose2d.hpp:21,
                 from simple_loop.cpp:16:
/opt/ros/indigo/include/ecl/linear_algebra/eigen.hpp:21:26: fatal error: Eigen/Cholesky: No such file or directory
 #include <Eigen/Cholesky>
                          ^
compilation terminated.
turtle1@turtle1:~/kobuki/src/kobuki_core/kobuki_driver/src/test$`

for Ubuntu 18.04, i guess it is understandable as it does not have kobuki core library suit, none the less, i will add it here: So I simple copied the simple_loop.cpp file to anyone of the packages got after following [This Link](https://github.com/gaunthan/Turtlebot2-On-Melodic) Which basically summarizes this post.

`/catkin_ws/src/kobuki_node/src/nodelet$ gcc -o sim simple_loop.cpp 
simple_loop.cpp:20:10: fatal error: ecl/time.hpp: No such file or directory
 #include <ecl/time.hpp>
          ^~~~~~~~~~~~~~
compilation terminated.
`

stonier, 2020-08-24:
Intent was to copy the file across and compile, not entire repo. 

You need to either add it to your cmake (easiest way) or make sure you point `gcc` at where the headers are with `-I<path_to_headers>`.

alexantosh, 2020-08-25:
Hello. I simply copied the simple_loop.cpp. I already had a similar kobuki repo on the indigo machine. 
Any way just to make sure. I copied the file again on the melodic machine and simple added it to the cmake as suggested. This is the error I got.
`n file included from /opt/ros/melodic/include/ecl/linear_algebra.hpp:19:0,
                 from /opt/ros/melodic/include/ecl/geometry/legacy_pose2d.hpp:21,
                 from /home/alex/catkin_ws/src/kobuki_keyop/src/simple_loop.cpp:22:
/opt/ros/melodic/include/ecl/linear_algebra/eigen.hpp:21:10: fatal error: Eigen/Cholesky: No such file or directory
 #include <Eigen/Cholesky>
          ^~~~~~~~~~~~~~~~
compilation terminated.
kobuki_keyop/src/CMakeFiles/keyop.dir/build.make:110: recipe for target 'kobuki_keyop/src/CMakeFiles/keyop.dir/simple_loop.cpp.o' failed
make[2]: *** [kobuki_keyop/src/CMakeFiles/keyop.dir/simple_loop.cpp.o] Error 1
CMakeFiles/Makefile2:3527: recipe for target 'kobuki_keyop/src/CMakeFiles/keyop.dir/all' failed
make[1]: *** [kobuki_keyop/src/CMakeFiles/keyop.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
`
This is in my catkin workspace trying to do a catkin_make

stonier, 2020-08-27:
I suspect you missed adding dependencies to the target in cmake.

alexantosh, 2020-09-03:
Hello, sorry for the tardy reply. did not have access to the desktop for a while. Apparently the issue was that my system has `eigen3` so I had to include that in the headers that required Eigen (I wonder if there's a simpler way of doing this instead of having to add it manually to all requiring files). Anyway it compiled finally but with warnings and failed because multiple mains were defined. Below is part of the log and the list of undefined references goes on. How effectively can I add these dependencies, again I apologise for the trouble:
`/opt/ros/melodic/include/kobuki_driver/kobuki.hpp:96:37: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
   void init(Parameters &parameters) throw (ecl::StandardException);
                                     ^~~~~
CMakeFiles/keyop.dir/simple_loop.cpp.o: In function `main':
simple_loop.cpp:(.text+0x11): multiple definition of `main'
CMakeFiles/keyop.dir/main.cpp.o:main.cpp:(.text+0x0): first defined here
CMakeFiles/keyop.dir/simple_loop.cpp.o: In function `kobuki::Parameters::Parameters()':
simple_loop.cpp:(.text._ZN6kobuki10ParametersC2Ev[_ZN6kobuki10ParametersC5Ev]+0x98): undefined reference to `kobuki::Battery::capacity'
simple_loop.cpp:(.text._ZN6kobuki10ParametersC2Ev[_ZN6kobuki10ParametersC5Ev]+0xa9): undefined reference to `kobuki::Battery::low'
`