yujinrobot/kobuki #388: Impossible to have in same /jointstates topic Wheels and P&Tilt !!

elpimous, 2018-03-25:
Hello, I encounter a problem with my kobuki robot (with pan and tilt head)
The kobuki works perfectly !! Thanks for this perfect robot !! And the docking station !!

But, when launching kobuki_node minimal.launch,
and same time dynamixels.launch,

I receive this topic : 
```
---
header: 
  seq: 1207
  stamp: 
    secs: 1522001104
    nsecs: 870637822
  frame_id: ''
name: [wheel_left_joint, wheel_right_joint]
position: [0.0, 0.0]
velocity: [0.0, 0.0]
effort: [0.0, 0.0]
---
header: 
  seq: 218
  stamp: 
    secs: 1522001104
    nsecs: 878241062
  frame_id: "base_link"
name: [head_pan_joint, head_tilt_joint]
position: [0.0, 0.0]
velocity: [0.0, 0.0]
effort: [0.0, 0.0]
---
```
I use kobuki stock apps,
here is my dynamixel.launch : 

```
<launch>
   <!-- Start the Dynamixel low-level driver manager with parameters -->
   <node name="dynamixel_manager" pkg="dynamixel_controllers"
      type="controller_manager.py" required="true" output="screen">
      <rosparam>
         namespace: alfred_dynamixel_manager
         serial_ports:
            dynamixel_ax12:
               port_name: /dev/ttyACM0
               baud_rate: 1000000
               min_motor_id:  1
               max_motor_id: 4
               update_rate: 20
      </rosparam>
   </node>
   
    <!-- Load the joint controller configuration from a YAML file -->
    <rosparam file="$(find alfred_launcher)/config/dynamixel_params.yaml" command="load"/>

   <!-- Start the head pan and tilt controllers -->
   <node name="dynamixel_controller_spawner_ax12" pkg="dynamixel_controllers"
   type="controller_spawner.py"
        args="--manager=alfred_dynamixel_manager
              --port=dynamixel_ax12
              --type=simple
        head_pan_joint
        head_tilt_joint"
        output="screen" />

   <!-- Start the Dynamixel Joint States Publisher -->
   <node name="dynamixel_joint_states_publisher" pkg="alfred_launcher" type="dynamixel_joint_state_publisher.py" output="screen" />
```

elpimous, 2018-04-01:
hello Stonier, I tried without success !!

I changed namespace in dynamixel.launch (see highter)
namespace: kobuki

I don't see errors, but I can't see both kobuki and p&tilt motors in same /point_states publish.

I'd like to have this topic : `rostopic echo /joint_states`
`name: [wheel_left_joint, wheel_right_joint, head_pan_joint, head_tilt_joint]`

And, regarding upper config files, why I have `frame_id: ''` ? 
I didn't touch any kobuki files !

Could anyone help me, please ?
Thanks a lot.
Vincent

stonier, 2018-04-02:
Kobuki's wheels are missing the parent frame_id (bug). But there is no code down from it that uses it yet either so hasn't been a problem to date.

Namespacing - I mean namespacing your nodes so that your dynamixel joint_states appears on a namespaced topic. Use the `ns` attribute in roslaunch (see the [wiki](http://wiki.ros.org/roslaunch/XML/node).