yujinrobot/kobuki #392: encoder reading

Python43coder, 2018-06-22:
Hello,

I am trying to get encoder reading from the kobuki base, but i didn't find any solution yet. Is it possible to get encoder readings rather than odom data?

PhillipSchmidt2017, 2018-06-22:
Hi, look for a topic */joint_states, it has the encoder readings.

Python43coder, 2018-06-25:
joint_states doesn't give the encoder reading. I checked it. Is there any other way to get encoder values?

PhillipSchmidt2017, 2018-06-25:
The topic has the position. Do you need the increments (/high-low signal)?

Python43coder, 2018-06-25:
yes, it has the position which is same as what  i get in odom topic. I want to get the raw encoder reading. i.e. how many ticks is the motor reporting? Is their any way to get it?

PhillipSchmidt2017, 2018-06-25:
I see, I'm wondering if the accuracy is sufficient to just divide tick_to_rad(0.002436916871363930187454f) [https://github.com/yujinrobot/kobuki_core/blob/devel/kobuki_driver/src/driver/diff_drive.cpp].
It seems that currently there is no interface for the encoder ticks (neither in ROS nor C++):
https://github.com/yujinrobot/kobuki_core/blob/devel/kobuki_driver/include/kobuki_driver/modules/diff_drive.hpp

PhillipSchmidt2017, 2018-06-25:
Yes, except that I am not aware that there are joint position in the odometry topic (http://docs.ros.org/kinetic/api/nav_msgs/html/msg/Odometry.html):
std_msgs/Header header
string child_frame_id
geometry_msgs/PoseWithCovariance pose
geometry_msgs/TwistWithCovariance twist
In the joint states there is
std_msgs/Header header
string child_frame_id
**geometry_msgs/PoseWithCovariance pose**
geometry_msgs/TwistWithCovariance twist
(http://docs.ros.org/kinetic/api/sensor_msgs/html/msg/JointState.html)

Python43coder, 2018-06-25:
thanks. I will look into it.