yujinrobot/kobuki #395: sensors/core at only 20hz

raywang-nx, 2018-12-04:
Ubuntu 16.04 and just install kobuki's necessary package, but find that this topic only publish at 20Hz while it should be 50 Hz. I didn't find the parameters to change the frequency, does anyone meet this before?
rostopic hz /mobile_base/sensors/core
subscribed to [/mobile_base/sensors/core]
average rate: 20.261
        min: 0.022s max: 0.060s std dev: 0.01632s window: 20
average rate: 20.160
        min: 0.022s max: 0.060s std dev: 0.01608s window: 40
average rate: 20.301
        min: 0.022s max: 0.060s std dev: 0.01618s window: 50

xjtao1314, 2018-12-15:
Hi
   I get the same problem, but I thought it may caused by the protocol have much more information than the serial can bear with speed of 115200. Btw , my hobuki have another strange behavior. I send fixed pure rotation command (1rad/s) , It will rapidly move to a rotation of 0.7rad/s ,and quite slowly reached 1.0 rad/s.  Do you get the same phenomenon ?  is it caused by the poor PID params?

raywang-nx, 2018-12-17:
@xjtao1314 
hi, i don't meet this problem, i just turn it faster a little every time due to the key_teleop. maybe when i set it to the fixed rotation i will come back to this issue again. 
as for the 20hz, it is 115200 bit/second, if the message is 200 bytes, then 200 * 8 * 50 is still smaller than 115200, while maybe the kobuki's logic is just limit the rate of this topic for some reasons

raywang-nx, 2019-01-14:
@xjtao1314 
hi, i think i find something. if i use USB to get the data, it will return at 20hz, but if i use serial port, i can get the data at 50hz. you can try this at kobuki_node\param\base.yaml: device_port: /dev/kobuki, and try /dev/ttyUSB0 or something else

closed 2019-01-14 by raywang-nx