yujinrobot/kobuki #399: Missing installation guide for ROS-melodic

zhaxiche, 2019-10-28:
I'm trying to install the kobuki stuff using ROS melodic. (http://wiki.ros.org/kobuki/Tutorials/Installation) The hardware is a Qbot2e running Ubuntu MATE 18.04 with RaspberryPi 3+. However, the installation guide for Melodic is missing. Could you please update it?

swinterbotix, 2020-02-20:
@zhaxiche, 

Many of the kobuki ROS packages don't seem to be officially supported by ROS Melodic just yet. You can see more at https://github.com/turtlebot/turtlebot/issues/272 regarding the turtlebot2 robot which depends on kobuki. It sounds like the only way to get it to work currently is to build from source. However, like you, I'll be very happy when the ros-melodic-kobuki-node package becomes available on apt-get.

nathangeorge1, 2020-03-20:
@swinterbotix I tried to install from source using https://github.com/gaunthan/Turtlebot2-On-Melodic/issues/5, but everything launches except GazeboRos::readCommonParameter, GazeboRosKobuki::preparePublishTf, or GazeboRosKobuki::setupRosApi which is necessary for the odom and tf data. Any workarounds?

swinterbotix, 2020-03-20:
I've got no idea - but thanks for pointing out that repo! I'll take a look at it at some point.

lukelu0520, 2020-08-02:
Ubuntu 16.04 + Kinetic would end support in April, 2021, now Kinetic version isn't recommended for new installs by ROS wiki.
So hope to find anyone who is running kobuki on Ubuntu 18.04 + Melodic. (or Ubuntu 20.04 + Neotic)

swinterbotix, 2020-08-13:
For anyone who is interested, I've attached an installation script that should get all the packages listed in the melodic branch of this repo working on Ubuntu 18.04 (especially kobuki_node) except for the kobuki_capabilities package (which doesn't seem to be that important anyway). To do this, I literally went through all the dependencies in those packages and confirmed that they would either by installed automatically by rosdep or are already present in the repo. Note that I've only confirmed that everything builds. Unfortunately I won't be able to test until tomorrow whether or not it runs properly since my Kobuki base is dead, and my charger is at work :(

[kobuki_install.zip](https://github.com/yujinrobot/kobuki/files/5067065/kobuki_install.zip)

lukelu0520, 2020-08-13:
Thanks for your contribution, @swinterbotix !

When you got chance, hope to try roslaunch kobuki_node minimal.launch to see if the error message "Malformed sub-payload detected" is gone. This issue ( #382 ) is kind of a mystery in 18.04 melodic.

Best,
-- Luke

swinterbotix, 2020-08-13:
I can definitely check for that. But the error you're talking about also happens in kinetic. From my experience (both personally and doing research online), it doesn't affect anything and should just be ignored.

lukelu0520, 2020-08-13:
@swinterbotix , thanks for your prompt reply.

It's pretty annoying so really want to know the root cause. We are also digging deeper to figure it out (for both curiosity and robotic health, a useful ref: http://yujinrobot.github.io/kobuki/enAppendixProtocolSpecification.html from Daniel) 

-- Luke

swinterbotix, 2020-08-13:
So I tested the robot today with the installation script by running the minimal.launch launch file. Everything seems to work just fine, but I do get the following warnings...

[image]

So I understand the first warning - about no robot description being found as the minimal launch file doesn't set the robot_description parameter. However, if I run the 'robot_with_tf' launch file which _does_ load the robot_description parameter, that warning goes away.

About the other warnings - it seems to say that the kobuki base firmware is outdated and should be upgraded. Unfortunately, I was not successful in doing this as their download links don't work. I created an Issue about it - #420. 

That said, the base still seems to function properly!

lukelu0520, 2020-08-13:
@swinterbotix , I got similar issues as you. Feel free to check #418 and regards to your 

> [ WARN] [1596651469.513160176]: Kobuki : no robot description given on the parameter server 

This one could be resolved by adding robot description in minimal.launch file. ref: kobuki_description/launch/view_model.launch
(from #382)

-- Luke

swinterbotix, 2020-08-13:
You're awesome @lukelu0520!

Firmware is now updated and warnings have disappeared!

It's interesting though how running minimal.launch in kinetic did not give the robot description warning though.

In any event, even with the updated firmware, I'm still seeing that sub-payload error message.

stonier, 2020-08-31:
If you'd like to help out rather than hack (yes, I know, hacking is fun) around the edges, delve into #422 :)

sridharsidhu0801, 2021-08-13:
> I'm trying to install the kobuki stuff using ROS melodic. (http://wiki.ros.org/kobuki/Tutorials/Installation) The hardware is a Qbot2e running Ubuntu MATE 18.04 with RaspberryPi 3+. However, the installation guide for Melodic is missing. Could you please update it?

@zhaxiche I am working on QBot 2e as well using ROS Melodic. I end up with some errors as well (mentioned below). Please let me know if you know the solution to this error. Also, let's collaborate and have proper communication to help each other in working with QBot 2e using ROS-Melodic.
Error as follows

ubuntu@ubuntu:~/catkin_ws$ catkin_make -j1
Base path: /home/ubuntu/catkin_ws
Source space: /home/ubuntu/catkin_ws/src
Build space: /home/ubuntu/catkin_ws/build
Devel space: /home/ubuntu/catkin_ws/devel
Install space: /home/ubuntu/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/ubuntu/catkin_ws/build"
####
####
#### Running command: "make -j1" in "/home/ubuntu/catkin_ws/build"
####
[  0%] Built target _kobuki_msgs_generate_messages_check_deps_ScanAngle
[  0%] Built target _kobuki_msgs_generate_messages_check_deps_DigitalInputEvent
[  0%] Built target _kobuki_msgs_generate_messages_check_deps_DigitalOutput
[  0%] Built target _kobuki_msgs_generate_messages_check_deps_ButtonEvent
[  0%] Built target actionlib_msgs_generate_messages_py
[  0%] Built target _kobuki_msgs_generate_messages_check_deps_Led
[  0%] Built target _kobuki_msgs_generate_messages_check_deps_KeyboardInput
[  0%] Built target _kobuki_msgs_generate_messages_check_deps_AutoDockingResult
[  0%] Built target _kobuki_msgs_generate_messages_check_deps_AutoDockingActionResult
[  0%] Built target _kobuki_msgs_generate_messages_check_deps_ControllerInfo
[  0%] Built target _kobuki_msgs_generate_messages_check_deps_WheelDropEvent
[  0%] Built target _kobuki_msgs_generate_messages_check_deps_BumperEvent
[  0%] Built target _kobuki_msgs_generate_messages_check_deps_AutoDockingActionFeedback
[  0%] Built target _kobuki_msgs_generate_messages_check_deps_CliffEvent
[  0%] Built target _kobuki_msgs_generate_messages_check_deps_AutoDockingFeedback
[  0%] Built target _kobuki_msgs_generate_messages_check_deps_AutoDockingAction
[  0%] Built target _kobuki_msgs_generate_messages_check_deps_SensorState
[  0%] Built target std_msgs_generate_messages_py
[  0%] Built target _kobuki_msgs_generate_messages_check_deps_AutoDockingActionGoal
[  0%] Built target _kobuki_msgs_generate_messages_check_deps_RobotStateEvent
[  0%] Built target _kobuki_msgs_generate_messages_check_deps_ExternalPower
[  0%] Built target _kobuki_msgs_generate_messages_check_deps_Sound
[  0%] Built target _kobuki_msgs_generate_messages_check_deps_DockInfraRed
[  0%] Built target _kobuki_msgs_generate_messages_check_deps_AutoDockingGoal
[  0%] Built target _kobuki_msgs_generate_messages_check_deps_MotorPower
[  0%] Built target _kobuki_msgs_generate_messages_check_deps_VersionInfo
[  0%] Built target _kobuki_msgs_generate_messages_check_deps_PowerSystemEvent
[ 18%] Built target kobuki_msgs_generate_messages_py
[ 18%] Built target actionlib_msgs_generate_messages_nodejs
[ 18%] Built target std_msgs_generate_messages_nodejs
[ 35%] Built target kobuki_msgs_generate_messages_nodejs
[ 35%] Built target std_msgs_generate_messages_eus
[ 35%] Built target actionlib_msgs_generate_messages_eus
[ 52%] Built target kobuki_msgs_generate_messages_eus
[ 52%] Built target actionlib_msgs_generate_messages_cpp
[ 52%] Built target std_msgs_generate_messages_cpp
[ 69%] Built target kobuki_msgs_generate_messages_cpp
[ 69%] Built target actionlib_msgs_generate_messages_lisp
[ 69%] Built target std_msgs_generate_messages_lisp
[ 87%] Built target kobuki_msgs_generate_messages_lisp
[ 87%] Built target kobuki_msgs_generate_messages
[ 87%] Built target rosgraph_msgs_generate_messages_cpp
[ 87%] Built target rosgraph_msgs_generate_messages_py
[ 87%] Built target rosgraph_msgs_generate_messages_nodejs
[ 87%] Built target rosgraph_msgs_generate_messages_lisp
[ 87%] Built target rosgraph_msgs_generate_messages_eus
[ 87%] Built target std_srvs_generate_messages_cpp
[ 87%] Built target geometry_msgs_generate_messages_py
[ 87%] Built target geometry_msgs_generate_messages_nodejs
[ 87%] Built target roscpp_generate_messages_lisp
[ 87%] Built target geometry_msgs_generate_messages_cpp
[ 87%] Built target roscpp_generate_messages_cpp
[ 87%] Built target geometry_msgs_generate_messages_lisp
[... 41 lines cut ...]
[ 94%] Built target actionlib_generate_messages_lisp
[ 94%] Built target actionlib_generate_messages_nodejs
[ 94%] Built target nav_msgs_generate_messages_lisp
[ 94%] Built target tf_generate_messages_lisp
[ 94%] Built target tf2_msgs_generate_messages_eus
[ 94%] Built target tf_generate_messages_nodejs
[ 94%] Built target tf2_msgs_generate_messages_nodejs
[ 94%] Built target actionlib_generate_messages_py
[ 94%] Built target nav_msgs_generate_messages_eus
[ 94%] Built target tf2_msgs_generate_messages_lisp
[ 94%] Built target tf2_msgs_generate_messages_cpp
[ 94%] Built target diagnostic_msgs_generate_messages_eus
[ 94%] Built target diagnostic_msgs_generate_messages_cpp
[ 95%] Building CXX object kobuki_node/src/library/CMakeFiles/kobuki_ros.dir/kobuki_ros.cpp.o
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
kobuki_node/src/library/CMakeFiles/kobuki_ros.dir/build.make:86: recipe for target 'kobuki_node/src/library/CMakeFiles/kobuki_ros.dir/kobuki_ros.cpp.o' failed
make[2]: *** [kobuki_node/src/library/CMakeFiles/kobuki_ros.dir/kobuki_ros.cpp.o] Error 4
CMakeFiles/Makefile2:4863: recipe for target 'kobuki_node/src/library/CMakeFiles/kobuki_ros.dir/all' failed
make[1]: *** [kobuki_node/src/library/CMakeFiles/kobuki_ros.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j1" failed

xkaraman, 2022-07-21:
> how can I have this working in gazebo. I installed everything as mentioned above. later on cleaned it and installed it by running the convenient script provided at [turtblebot2onMelodic](https://github.com/gaunthan/Turtlebot2-On-Melodic). When i ran the gazebo simulation in the empty world, it runs fine. but teleoperation or directly publish twist objects the the cmd_vel_mux/ topics does not work. At least the robot is not moving. how can i achieve this for simulation purposes? thanks

I know this is old topic but for anyone still having issues, I believe `kobuki_desktop` package is necessary for simulations purposes. Build it from source, as no melodic package is yet out. `kobuki_qtestsuite` can be `catkin_ignore`d since it fails to build for me and it's for testing.

After building and sourcing the package, you can teleop successfully in the simulation environment!