yujinrobot/kobuki #400: description: wheel_radius(left) != wheel_radius(right) ?

gavanderhoorn, 2019-11-25:
Not sure this matters, but it would appear the `wheel_left_link` and `wheel_right_link` have slightly different radii configured for the `collision` geometry.

Left:

https://github.com/yujinrobot/kobuki/blob/161f69310ef94d2ab7566cf085231e832b8985cf/kobuki_description/urdf/kobuki.urdf.xacro#L78

Right:

https://github.com/yujinrobot/kobuki/blob/161f69310ef94d2ab7566cf085231e832b8985cf/kobuki_description/urdf/kobuki.urdf.xacro#L106

Is this on purpose?

corot, 2019-11-26:
mmmm......  never noticed this. I guess no.

gavanderhoorn, 2019-11-26:
Do we know which one is the correct radius?

See also [kobuki don't go straight, get a little to the right in gazebo](https://answers.ros.org/question/338481) on ROS Answers.

corot, 2019-11-26:
According to
https://yujinrobot.github.io/kobuki/enAppendixKobukiParameters.html, 35 mm

wow!  I was considering if that error could have any impact on Gazebo!
very good catch!

gavanderhoorn, 2019-11-26:
Will you PR a fix, or should 'someone else' do that? :)

commit f2d83ce4 (2019-11-28): Fix #400: set both wheels to 35mm

Thanks to @gavanderhoorn  for the catch!
closed 2019-11-28 by corot