yujinrobot/kobuki #408: Unable to build from source after 'eloquent message variable name refactor'

ljazzal, 2020-01-24:
Hello,

I ran into some issues recently while building kobuki from source after the message variables have been refactored in [commit](https://github.com/yujinrobot/kobuki/commit/b609a185e249d127679e7b18a24d606acbdf918f):

`test_ws/src/kobuki/kobuki_keyop/src/keyop_core.cpp:274:28: error: ‘const KeyboardInput {aka const struct kobuki_msgs::KeyboardInput_<std::allocator<void> >}’ has no member named ‘pressed_key’; did you mean ‘pressedKey’?
   processKeyboardInput(key.pressed_key);
                            ^~~~~~~~~~~
                            pressedKey
`
`test_ws/src/kobuki/kobuki_keyop/src/keyop_core.cpp:293:38: error: ‘KEYCODE_LEFT’ is not a member of ‘kobuki_msgs::KeyboardInput {aka kobuki_msgs::KeyboardInput_<std::allocator<void> >}’
     case kobuki_msgs::KeyboardInput::KEYCODE_LEFT:
`

Reverting to [f2d83ce467ff52c6c20fefaf0a54d3464e3a212e](https://github.com/yujinrobot/kobuki/commit/f2d83ce467ff52c6c20fefaf0a54d3464e3a212e) fixed the issue!

I don't know if this expected given the port over to ROS2 and the legacy status of kobuki_msgs

stonier, 2020-01-24:
I am guessing you are building on a ROS1 distribution?

All the sources should still be valid, repos unchanged, just make sure you don't use `devel` branches. There are no promises attached to the devel branch (naturally, otherwise where would you engage in development!). Use the release branches instead - e.g. [kobuki/melodic](https://github.com/yujinrobot/kobuki/tree/melodic). Each repository has similar branches and will automatically aggregate release bugfixes for free.

Otherwise, pick a tag, e.g. [0.7.6](https://github.com/yujinrobot/kobuki/tree/0.7.6). At least with these, they come with a promise that it *should* work.

stonier, 2020-01-24:
Note, in the near future this repo too (like `kobuki_core` and `kobuki_msgs`) will lose the devel branch and default to `melodic` with a better README clarifying the situation and pointing to the ros2 repos. You just happen to have caught us in the middle of these changes.

ljazzal, 2020-01-24:
That makes sense. Thanks for the clarification!

Feel free to close this as this isn't really an issue

closed 2020-01-24 by stonier