yujinrobot/kobuki #415: error: ‘KEYCODE_RIGHT’ is not a member of ‘kobuki_msgs::KeyboardInput {aka kobuki_msgs::KeyboardInput_<std::allocator<void> >}’      case kobuki_msgs::KeyboardInput::KEYCODE_RIGHT:

1046652786, 2020-05-14:
i had changed the package name ,before i changed it was named kobuki_keyop, how can i solve it?

antondim, 2020-05-20:
Had the same problem. Replace the false values with the ones stated in http://docs.ros.org/hydro/api/kobuki_msgs/html/msg/KeyboardInput.html

For example, "KEYCODE_RIGHT" changes into "KeyCode_Right" etc etc.

Those values can be found in the source cpp file "keyop_core.cpp". 

Hope this helps.

corot, 2020-05-20:
I changed kobuki_msgs/devel to match changes on devel here: b609a185e249d127679e7b18a24d606acbdf918f
So compiling latest devel together should work.

What are you trying to compile?

rjam92, 2020-08-19:
Was this problem resolved? what was the installation process other than cloning this in catkin_ws/src and building it?

stonier, 2020-08-19:
Problem I'm guessing is that you're using `devel` branch of `kobuki_keyop` which was part of an aborted ros2 migration.You should be using the release branch for melodic, which was `release/0.7-melodic` but it looks like @corot has also been working on `melodic`.

rjam92, 2020-08-20:
Just to make sure I am clear about what you are saying:
1. I should clone and build the melodic branch instead of the default devel branch 
2. Transfer any relevant fixes from devel branch to the melodic branch

If not so can you explain in more detail how to carry out the above steps? 
Thank you

stonier, 2020-08-21:
I'll send some PR's that cleanup the branch situation ...

-----

The connection problem sounds orthogonal.

* What do you get when you type `groups`? You usually have to log out/reboot to have the addition to the `dialout` group take affect.
* Do you see anything come up on `/dev/ttyUSB0` when you plug in kobuki?