yujinrobot/kobuki #418: Can not download the The Flashing Utility stm32flash-0.4.tar.gz

lukelu0520, 2020-08-05:
Dear kobuki friends,

Tried by multiple computers, but we can not connect to the file server (http://files.yujinrobot.com/kobuki/firmware/stm32flash-0.4.tar.gz) to download the flashing utility. 

Was wondering if someone has the same issue. Hope it is not permanent down...  
Our location is in California, US.

-- Luke

stonier, 2020-08-05:
You can find it at https://sourceforge.net/projects/stm32flash/files/.

Still a WIP, but rebuilding documentation while doing a ROS2 port - you can find updated notes at https://kobuki.readthedocs.io/en/devel/firmware.html#updating-firmware. Let me know if you run into any problems.

lukelu0520, 2020-08-05:
Thanks, @stonier ! Updated !

closed 2020-08-05 by lukelu0520