yujinrobot/kobuki #419: No response on kobuki_testsuite test_rotation.py and test_translation.py

lukelu0520, 2020-08-05:
Dear kobuki friends,

We went through testsuite and everything works great except below two tests:
> rosrun kobuki_testsuite test_rotation.py
rosrun kobuki_testsuite test_translation.py

Above two tests have no response. 

We also tested _roslaunch kobuki_keyop safe_keyop.launch --screen_ , the control works great. So were wondering why simpler test_rotation.py and test_translation.py don't work.

Hope to hear your expertise.

-- Luke

corot, 2020-08-06:
I did that test suite like three lifes ago... :neckbeard: 
If I remember properly, we tested rotation and translation by placing a calibration checkboard pattern on top of Kobuki and monitoring its movements with a camera placed above. So the scripts you mention just make the robot move; you need the camera setup to check its correctness.

130s, 2023-05-08:
Another user may have a similar/same question [answers.ros.org#415112](https://answers.ros.org/question/415112).