yujinrobot/kobuki #423: melodic release for apt install in Ubuntu 18

hanzheteng, 2020-09-09:
Hi,

I am wondering if there is a melodic release for `kobuki_description` package in Ubuntu 18, such that I can install it by `sudo apt install ros-melodic-kobuki-description`. I see the meta-package `ros-melodic-kobuki` is still only available in Ubuntu 16 and ROS Kinetic right now. 

Our customized Turtlebot2 robot is built on top of kobuki base. If `kobuki_description` is available, we can then use `wstool` and `rosinstall` to automate the installation, otherwise it is a bit cumbersome for our users to build everything from source.

swiz23, 2021-01-07:
I've run into this issue as well. Currently, the way I get around this is by writing my own bash script to install my robot software. Part of the install script is to clone the repo from source, checkout the melodic branch, delete the `kobuki` metapacakge and the `kobuki_capabilities` ros package (as rosdep can't build these due to lack of dependencies), and then build the repo.

hanzheteng, 2021-02-19:
Thanks for the info. Just FYI, the script we use to install the customized Turtlebot2 in Ubuntu 18 can be found [here](https://github.com/UCR-Robotics/Turtlebot2-On-Melodic/blob/master/install_turtlebot.sh).