yujinrobot/kobuki #427: Is a release for ROS Noetic planned anytime soon?

rjam-xav, 2020-12-01:
Hello,
Is there a plan to release the packages for ROS noetic as well?

swiz23, 2021-02-15:
Looks like there still is some work to be done. However, if all you care about is getting this repo to compile, I have been able to get it working on ROS Noetic by building from source.

Clone this repo into your workspace. You'll also want to clone https://github.com/yujinrobot/yujin_ocs into your workspace as well. Many of the directories in the yujin_ocs arn't necessary. Delete everything except 'yocs_cmd_vel_mux', 'yocs_controllers', and 'yocs_velocity_smoother'.

If it's not already installed, you'll want to install `liborocos-kdl-dev`. Apparently in ROS Noetic, the `python-orocos-kdl` ROS package does not exist since it was decided to depend on the system version instead. It's needed as a dependency for the kobuki_testsuite package. So if you run the `rosdep install --from-paths src --ignore-src -r -y` in the root of the workspace, you'll see an error message saying that rosdep can't find the `python-orocos-kdl` package. This can be ignored though since the dependency is already satisfied with `liborocos-kdl-dev` despite ROS not knowing about it.

martinerk0, 2022-06-22:
I created shell script for installation per @swiz23 reply.
Tested on Jetson Nano 20.04 with Noetic.
https://github.com/martinerk0/kobuki_noetic